thomasfredericks/Bounce2 #13: press and hold

eadf, 2015-02-20:
I needed two functions in one button so added a [```duration()``` method ](https://github.com/eadf/Bounce2) to your class.
It simply returns the period [ms] the pin has been in current state (totally undisturbed - flatline)

If you think this is something you would like in your library i can make a pull request.
Or you can add it yourself, it's effectively a one-liner.

thomasfredericks, 2015-02-23:
I will look into it, but if I recall correctly, there was an issue with the BOUNCE_LOCK-OUT mode.

eadf, 2015-02-24:
An alternative would be to keep another timer value that remembers when the output value of the debouncer changed. 

Perhaps that's more semantically correct, a user of the debouncer that wants to know if the button has been pressed for 1 seconds or more doesn't care if there were some jitter 10ms ago.

MrHaroldA, 2018-02-13:
Isn't this as simple as to expose `previous_millis` as public? In my code I have to declare and fill my own variable to store exactly the same data to check if a switch is pressed long.

Anyway: this would be a very welcome addition for me.

thomasfredericks, 2018-03-01:
> Isn't this as simple as to expose previous_millis as public? In my code I have to declare and fill my own variable to store exactly the same data to check if a switch is pressed long.

No. As @eadf notes, previous_millis is an internal timer that does not necessarily match the duration of a press. 

If someone wants to code and test it, I will try to integrate it.

thomasfredericks, 2018-03-08:
Okay, in the new "duration" branch (https://github.com/thomasfredericks/Bounce2/tree/Duration) I added two methods:
* `duration()` that returns how long the current state has been lasting
* `previousDuration()` that returns how long the previous state has been lasting

What do you think?

DrFirexx, 2018-09-07:
Hi where is this previousDuration() I can't see it in the code

MrHaroldA, 2018-09-07:
Can't find it either; but thank you for the reminder to check this out ;)

DrFirexx, 2018-09-07:
Yeah I'm trying to implement a pressed button using duration() but can't get hold of it as it returns 0 whenever button move do you have any tip? I'm quite new to all of this.

thomasfredericks, 2018-09-11:
This was implemented in a branch. It was removed when the branch was merged for release as it was not sufficiently tested. I added a test ```previousMillis()``` in the master branch,

closed 2019-05-10 by thomasfredericks

thomasfredericks, 2021-04-01:
Yes, in the examples. https://github.com/thomasfredericks/Bounce2/tree/master/examples/more/duration

pReya, 2021-07-14:
> Yes, in the examples. https://github.com/thomasfredericks/Bounce2/tree/master/examples/more/duration

Somehow, the example called `duration.ino` actually doesn't use the `duration()` function. Is this on purpose?

thomasfredericks, 2021-07-18:
Oups. Sorry. You are right. The example needs to be updated. I will do it once I get back from vacation.

pReya, 2021-08-02:
@thomasfredericks Not trying to annoy you on vaction ;) But I'd really appreciate a better example.

thomasfredericks, 2021-08-03:
No worries @pReya , I uploaded 3 new duration examples in the "examples/more" folder: https://github.com/thomasfredericks/Bounce2/tree/master/examples/more/

thomasfredericks, 2021-08-03:
Please note I changed duration() to currentDuration() to avoid any confusion.
currentDuration() = duration of current state
previousDuration() = duration of previous state

pReya, 2021-08-04:
Thank you so much! Appreciate it a lot. I think it would be a good idea to add the term "long press" or "button hold" to the folders/filenames/readme, because it's a term that's googled a lot. Might bring new users to the library (if that's what you're after).