thomasfredericks/Bounce2 #19: Could you ask for inclusion of this library in Arduino's Board Manager?

q2dg, 2015-07-24:
This would give more visibility to this library, I think.
Thanks

thomasfredericks, 2015-07-24:
I guess you mean Arduino's Library Manager.
Sure, I will read about it.

rhettc, 2016-05-13:
looks closed to me

PaulStoffregen, 2016-05-13:
Looks like the only thing you need to do is open an issue on Arduino's tracker, asking them to add your library.

They do have some requirements (library.properties, tags, releases), but you've already done that work.

thomasfredericks, 2016-05-14:
Will do.

rhettc, 2016-05-14:
To clarify, this library is already included in the Arduino Library Manager. However, it's using the 2.1 tag, and we'd probably benefit from a new release.

thomasfredericks, 2016-05-14:
Ok, I made a new release. It should eventually update automatically I think.

rhettc, 2016-05-16:
Confirmed that the Library Manager has picked up on version 2.2.0 
Awesome

closed 2016-05-16 by thomasfredericks