thomasfredericks/Bounce2 #26: "_BV" doesn't compile. Changed to "bit" and works

softegg, 2016-06-26:
I'm compiling for Arduino Due, and it seems like there is no _BV macro anymore. I did a little research and determined that there is a "bit" macro with the same syntax that does the same thing, so I did a search and replace and that worked.

softegg, 2016-06-26:
I made a fix and submitted pull request #27.
https://github.com/thomasfredericks/Bounce2/pull/27

thomasfredericks, 2016-06-27:
Where is the _BV macro defined? Is this only with the Due? Are you using Arduino.cc or Arduino.org?

PaulStoffregen, 2016-06-27:
_BV() has been so widely used in Arduino code for so long, I had always assumed it was part of the Arduino code library.  But looking into this, it turns out to be part of avr-libc.

http://www.nongnu.org/avr-libc/user-manual/group__avr__sfr.html

Perhaps a simpler patch might look like:

    #ifndef _BV
    #define _BV(n) (1<<(n))
    #endif

softegg, 2016-06-27:
It builds on the Arduino Mega 2560. I'm guessing a problem with non AVR cores, based on Paul's comment.
I'm running Arduino 1.6.6. The "About Arduino" references Arduino.cc . 
I chose to replace every "_BV" with "bit" because "bit" is actually in Arduino.h, so presumably that is the party approved solution for Arduino.
Personally, I think typing out the bit shift code explicitly (1<<bits) is preferable, because nobody has to guess what it does, and it's about the same number of characters. I can't see any advantage to making a macro of this. But since it was done, I'm trying to stick with the system.

thomasfredericks, 2016-06-28:
I will add Paul's patch as it is the minimum of modification required to fix this.

closed 2016-06-28 by thomasfredericks

HamishB, 2016-10-25:
FYI this was also a problem with Cortex M0 SAMD21 (used by Arduino Zero and others like the Adafruit Feather M0 boards).  With the patch it builds fine.

regards,
Hamish