thomasfredericks/Bounce2 #32: WiringPi support

gamecompiler, 2017-08-08:
Hi, i would like to use Bounce2 with WiringPi.
But id got stock with the error:

> undefined reference to 'Bounce::attach (int)'
Here is my code for reference:

'''

/*This source code copyrighted by Lazy Foo' Productions (2004-2015)
and may not be redistributed without written permission.*/
/*
 *
 *08.08.2017 - Justus Metzler
 *	Der Code wurde so stark verändert das ich defintiv nicht Lazy Foo wegen den paar Zeilen eine Email schreibe.
 *	Auf jeden Fall basiert dieser Code auf Lazy Foos 21. Tut: http://lazyfoo.net/tutorials/SDL/21_sound_effects_and_music/index.php
 *	Mit einer Erwähnung von LazyFoos Tutorial ist glaube ich genüge getan.
 *
 *	Dieser Code spielt die Datei "/data/sounds/milshot.wav" ab sobald der GPIO 5 high wird (WiringPi gpio NICHT BCM), und das in einem externen thread. 
 *	Die delay Fkt im Code unten ist ekelhaft, aber da der Sound in einem seperaten thread abgespielt wird ist mir das in diesem Bsp absolut egal.
 *	Zudem ist es nur ein Test wie man SDL nutzt.
 *	
 *	Alleine die Installation von SDL ist eine Katastrophe auf dem RPi. Viel Glück/Spaß
 *
*/

//Include SDL2
#include <SDL.h>
#include <SDL_mixer.h>
#include <stdio.h>
#include <string>

//Include WiringPi
#include <wiringPi.h>

//Include Bounce2
#include <Bounce2.h>

#include <iostream>
using namespace std;

//Load SDL
bool init();

//Load Media
bool loadMedia();

//Clear
void close();

//Sounds
Mix_Chunk *milshot = NULL;

bool init()
{
	//Initialization flag
	bool success = true;

	//Initialize SDL
	if (SDL_Init(SDL_INIT_AUDIO) < 0)
	{
		printf("SDL could not initialize! SDL Error: %s\n", SDL_GetError());
		success = false;
	}
	
	//Initialize SDL_mixer
	if (Mix_OpenAudio(44100, MIX_DEFAULT_FORMAT, 2, 2048) < 0)
	{
		printf("SDL_mixer could not initialize! SDL_mixer Error: %s\n", Mix_GetError());
		success = false;
	}
	return success;
}

bool loadMedia()
{
	//Loading success flag
	bool success = true;

	//Load sound effects
	
	milshot = Mix_LoadWAV("../data/sounds/01_milshot.wav");
	if (milshot == NULL)
	{
		printf("Failed to load high sound effect! SDL_mixer Error: %s\n", Mix_GetError());
		success = false;
	}
	

	return success;
}

void close()
{

	//Free the sound effects
	Mix_FreeChunk(milshot);
	milshot = NULL;

	//Quit SDL subsystems
	Mix_Quit();
	SDL_Quit();
}

//Code snippets
/*
 *	
 *	Mix_PlayChannel(-1, milshot, 0); <== play soundfile in mixer
 *	digitalRead(pin); <== read pin
 *	cout << "milshot" << endl;	//<== print "milshot" to console
 */
//Game variables

uint8_t health = 255;
uint8_t shells = 12;
uint8_t bullets = 255;

uint8_t max_health = 255;
uint8_t max_shells = 12;
uint8_t max_bullets = 255;

uint8_t player_id = 1;
uint8_t team_id = 2;
uint16_t reloadtime = 2000;

Bounce debouncer1 = Bounce(); 
Bounce debouncer2 = Bounce(); 

void shot_handler()
{
	if (digitalRead(5))
	{
		
	}
}

int main(int argc, char* args[])
{
	
	//Start up SDL
	if (!init())
	{
		printf("Failed to initialize!\n");
	}
	else
	{
		//Load media
		if (!loadMedia())
		{
			printf("Failed to load media!\n");
		}
		else
		{	
			//Main loop flag
			bool quit = false;

			//Event handler
			SDL_Event e;
			
			//Setup wiringPi
			if (wiringPiSetup() == -1)
				return 1;
			
			
			
			pinMode(5, INPUT);
			debouncer1.attach(5);
			debouncer1.interval(5); // interval in ms
			
			pinMode(4, INPUT);
			debouncer2.attach(4);
			debouncer2.interval(5); // interval in ms
			
			
			
			//While application is running <<======================================================================================
			while (!quit)
			{
				debouncer1.update();
				debouncer2.update();
				
				// Get the updated value :
				int value1 = debouncer1.read();
				int value2 = debouncer2.read();

				  // Turn on the LED if either button is pressed :
				if (value1 == LOW || value2 == LOW) {
					cout << "ON" << endl;	//<== print "milshot" to console
				} 
				else {
					cout << "OFF" << endl;	//<== print "milshot" to console
				}
				
			}
			//<====================================================================================================================
		}
	}

	//Free resources and close SDL
	close();

	return 0;
}

'''

thomasfredericks, 2017-08-08:
I am sorry I do not know WiringPi. You will have to ask the WiringPi community. Good luck.

thomasfredericks, 2017-08-08:
If you do figure it out, please reply. Thank you.

gamecompiler, 2017-08-08:
Id got it working. Im an idiot.... Id forgot to add it the Bounce.cpp to the source files....
Bounce2 works with WiringPi!!! Thats awesome.

gamecompiler, 2017-08-08:
[WiringPi_Bounce2_SDL_example.zip](https://github.com/thomasfredericks/Bounce2/files/1209314/WiringPi_Bounce2_SDL_example.zip)
Heres the complete sketch with makefile. You may want to add it to the examples directory.

thomasfredericks, 2017-08-08:
Thank you. :thumbsup:

closed 2017-08-08 by thomasfredericks