thomasfredericks/Bounce2 #33: Small "Dead Zone"

gfvalvo, 2017-09-29:
Hi.

I just ran across your library. Nice, tight code.

I do see one minor issue related to Line 71 of ‘bounce2.cpp’: It happens if the switch state doesn’t change for a long time and millis() rollover occurs. Then, as millis() increases from 0 again, there will be a small dead zone between millis() = previous_millis and millis() = previous_millis + interval_millis where a change of state will not be promptly detected even though the interval period is long-expired. This dead zone will occur every ~1193 hours.

Not a huge deal as it’s a very small window of time, only interval_millis milliseconds long, every 1193 hours. It could be fixed by adding a flag to indicate that the timeout interval has expired.

Thanks.

thomasfredericks, 2017-09-29:
Hi, I can't find the issue, but this was already discussed. Oh, maybe not here but for the Chrono library. Anyways, the code ```(millis() - previous_millis >= interval_millis)``` should handle the overflow problem.

For example, lets say the rollover for millis() is 0 to 255. The previous recorded start time (previous_millis ) was 200 and we want an interval_millis of 70:

* Once millis() reaches 255, ```millis() - previous_millis``` returns 255-200 = 55
* Once millis() overflows and reaches 0, ```millis() - previous_millis``` returns 0-200 = -200 = 56 with overflow
* Once millis() reaches 14, ```millis() - previous_millis``` returns 14-200 = -186 = 70 with overflow and this will return true

gfvalvo, 2017-09-29:
OK, now continue your example a little further. Let’s say there’s still no switch state change as millis() continues to increase after rollover. Now millis() reaches 202 and the switch is pushed:

**millis() - previous_millis** returns **202-200 = 2** and the condition will return false even though 258 milliseconds have passed! Thus, the ‘if’ statement will fail and the new switch state won’t be promptly detected.

thomasfredericks, 2017-10-02:
In your latest example, this will only happen if your interval is longer than the overflow value which is not an issue with the real code unless your interval is many days. In my example, the state switch will happen once the counter reaches 14, counting from 200, for an interval of 70. Please give me a more complete example if I am missing something.

gfvalvo, 2017-10-02:
OK, keeping with your example of millis() running from 0 to 255 (and then rolling over), a previous_millis of 200 and an interval_mills of 70:

So, the millis() counter keeps incrementing from 200, reaches 255, rolls over to 0 and keeps counting. Imagine during this time there is no change in the input we're debouncing (switch is not pressed). All is well, millis() reaches 14 and our 70ms timing interval has "expired".

But, still, nobody has pressed the switch and millis() continues to count. Finally, when millis() reaches 202, somebody presses the switch. This change should be promptly recognized as it has been 258ms since previous_millis was set to 200 (256 + 2).

But, when Line 71 of Bounce2.cpp checks '**millis() - previous_millis**', it gets 2**02 - 200 = 2**. Thus, the 'if' condition fails. In fact, the change of state won't be recognized until millis() increments past 255, rolls over, and reaches 14 again. Meaning, instead of promptly detecting the state change, it's not recognized until 68ms later.

Does that make sense, or am I delusional?

thomasfredericks, 2017-10-03:
Ah ha. I finally understand. As it is such a rare occasion, it is not a priority for me. If you work on a fix, please send it. Thank you for taking the time and noting the issue.

gfvalvo, 2017-10-04:
Agreed, the "window of vulnerability" is only 'interval_mills' milliseconds wide (say 20mS) out of 2^32 mS. That’s less than 5 parts out of 10^9 !!!

I really only looked at it because I found your library and am trying to learn C++ after being a C coder for more than 30 years.

As an exercise, I did fix this issue. But, I probably ended up make more changes than you're going to like :)
Anyway, it's attached as a zip file.

Thanks for the Bounce2 library, feel free to close this issue.

[NewDebounceTest.zip](https://github.com/thomasfredericks/Bounce2/files/1354312/NewDebounceTest.zip)

gfvalvo, 2017-10-04:
While I was messing around with your code, I also made the debounce method selectable when the Bounce object is instantiated rather than having to change the .h file.
[NewDebounceTest.zip](https://github.com/thomasfredericks/Bounce2/files/1356018/NewDebounceTest.zip)

thomasfredericks, 2017-10-30:
Hi, thank you for your contributions. Your NewDebounceTest.zip code changes the code a bit too much for me to feel comfortable to merge the changes. But I have created a new banch called deadzone. 

As for Bounce3.zip I prefer not to include any changes to the public part of the code for now.

thomasfredericks, 2019-05-10:
While the library has changed since this issue was found, I would like to address it for the next release. I think it could be solved by replacing `millis() - previous_millis >= interval_millis` with a function that references a variable that keeps in memory if the interval ever overflowed.