thomasfredericks/Bounce2 #39: rose() and fell() do not work if called less frequent than update()

jokrug, 2018-01-03:
Common problem: 
  - User must press a button to trigger a function in the program
  - User must release the button and press it again to trigger the function another time

One could use the rose() or fell() - function for this, but they only work, if they are called in the same loop directly after update().
I my program update() is called often in loop(), but the rose() or fell() - functions are called much less frequent in a timer routine and do not work because the STATE_CHANGED-flag is reset at every call of update().
I moved the "state &= ~_BV(STATE_CHANGED);"-line from update() to rose() and fell(). Now the STATE_CHANGED-flag remains set until rose() or fell() is called the first time. 
This is in my opinion the behaviour, that most people would expect and therefore would like to make a pull-request.
A problem could be compatibility because after the modification rose() and fell() would return a different value if called twice without an update() between the calls. 
One could define new functions for the modified rose() and fell(), but I think this is not necessary and makes the class difficult to use. It would be hard to explain the difference between the old an new functions.

thomasfredericks, 2018-01-04:
Ideally, the reactivity of your user interface (or sensors) should be instantaneous. The recommended way is to process all de-bounced inputs at the start of the loop, storing in user variables if needed later. Checking the input inside a timer is not optimal design.

In your suggestion a previous change in the input would be stored way over the expected time frame resulting in unexpected triggers. For your suggestion to work, you would have to poll rose() or fell() continuously which brings us back to our initial problem.

PaulStoffregen, 2018-01-05:
Maybe this is an ok moment to bring up a feature I've been developing for Teensy?.  Hopefully this will someday be adopted by the Arduino devs, though realistically it can take months or even years to get them to buy into almost any idea.  For the near future this is likely to be only available on Teensy.

It's called EventResponder.  The idea is you can create an EventResponder object and tell it you want a function run when the event occurs.  There are a few different options for *how* the function gets run, but the default is it happens when loop() or delay() or other stuff calls yield().  EventResponder is mainly designed so you can safely trigger functions to run from inside interrupts like timers, and EventResponder provides a way to pass data to the called function.  Later the function is called from ordinary non-interrupt context, where it's safe to use String, Serial, and other libraries.

So far only a few libs have been edited to actually use this.  Bounce2 is high on my list of libraries to adapt for EventResponder.  I know there probably won't be much interest to merge this into the official Bounce2 repository while EventResponder remains a Teensy-only feature.  That's ok.  I just wanted to mention this idea now, in case there's any feedback or interest in how it develops.

My tentative plan involves a fork of Bounce2 with a subclassed object, maybe named "BounceEvent" which works the same as regular Bounce, but has OnRise(function) and OnFall(function) to establish the functions to be called.  A BounceEvent could then be used from a timer interrupt which just calls update(), and the functions will automatically be called by the EventResponder infrastructure in the core library.

jokrug, 2018-01-06:
Paul, the idea of EventResponders sounds good, but does not suit my current programs.
There it is important, that the reaction on button events only takes place in certain situations.

To come back to the initial issue: after playing around with the Bounce2-lib for a few hours, I came to some conclusions:
- My initial idea to move the unset of the STATE_CHANGED-flag to rose() and fall() was indeed not good. Tomas gave some reason and I can add, that you would only be able to react on rose() OR fall() in one loop, because the call of either function would reset the STATE_CHANED-flag and the call to the other function would always return false. 
- In my opinion the is an error in fell(): 
```c
bool Bounce::fell()
{
    return !( state & _BV(DEBOUNCED_STATE) ) && ( state & _BV(STATE_CHANGED));
}
```
is equivalent to 
return !rose();

The fell() - function returns always true, if the DEBOUNCED_STATE is false or STATE_CHANGED is false or both. It should look like this:
```c
bool Bounce::fell()
{
    return ( !(state & _BV(DEBOUNCED_STATE) ) && ( state & _BV(STATE_CHANGED)) );
}
```
I wrote a little test sketch for tests. Try it yourself:
```c
#include <Bounce2.h>

#define GPIO_TASTE   10 

Bounce  deBouncedButton = Bounce();

void setup() {
  // put your setup code here, to run once:
  deBouncedButton.attach(GPIO_TASTE, INPUT_PULLUP);
  deBouncedButton.interval(10);
  Serial.begin(115200);
  Serial.println("Setup done");
}

void loop() {
  if( deBouncedButton.update() )
  {
    Serial.print("update() ");
  }
  if( deBouncedButton.rose() )
  {
    Serial.print("rose() state ");
    Serial.println(deBouncedButton.read());
  }
  if( deBouncedButton.fell() )
  {
    Serial.print("fell() state ");
    Serial.println(deBouncedButton.read());
  }
}
```

thomasfredericks, 2018-01-06:
@PaulStoffregen We have been working on an object-oriented Arduino library for creative physical computing called Plaquette. Documentation here : https://sofapirate.github.io/Plaquette/ 
It is more flow based but due to its structure it supports events and allows simple code like this:
```
// BLINKING LED
#include <Plaquette.h>
DigitalOut myLed(13);
SquareOsc myOsc(2.0, 0.5);

void begin() {}

void step() {
  myOsc >> myLed;
}
```

```
// BUTTON CONTROLS A LED
#include <Plaquette.h>

DigitalOut led; // The LED.

DigitalIn button(2); // The button, defaults to INTERNAL_PULLUP

void begin() {
  button.smooth(); // debounce button
}

void step() {
  button >> led; // Send button value to LED.
}
```

@jokrug I will test your error report with the _BV replacement pull by NelsonVides.

thomasfredericks, 2018-01-10:
Fixed.

closed 2018-01-10 by thomasfredericks