thomasfredericks/Bounce2 #49: error: invalid conversion from 'int' to 'WiringPinMode' [-fpermissive]

elmarfaber, 2018-10-18:
Hello,
I try to use the bounce2 under STM32 arduino code and get the following error:
.platformio\lib\Bounce2_ID1106\src/Bounce2.h:188:26: error: invalid conversion from 'int' to 'WiringPinMode' [-fpermissive]
I use VisualStudioCode with the PlatformIO extension.

Best regards
Elmar

thomasfredericks, 2018-10-22:
Hum, it seems that the platform you are using defines pin modes as an enum. 

This is a similar issue: https://www.stm32duino.com/viewtopic.php?t=463&start=20

I do not use PlatformIO yet. Sorry I can not help with this right now.

krukhlis, 2019-04-18:
> invalid conversion from 'int' to 'WiringPinMode' [-fpermissive]
This is not VS/PlatformIO issue. This is result of conversion of the Arduino Wire library to STM32 ( some company originally, stm32duino nowadays).
In original Arduino function is declared as:
`void pinMode(uint8_t pin, uint8_t mode)
`Whilst in stm32 version ( [STM32 io.h](https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/master/STM32F1/cores/maple/io.h) ) it is defined for Hell-known reasons as:
`void pinMode(uint8 pin, WiringPinMode mode) 
`Obviously code in Bounce2 targets traditional Arduino Wire API. 
For your issue -- just copy  void pinMode(uint8 pin, WiringPinMode mode)  implementation, but change param to uint8_t and do corresponding changes in implementation to parse byte instead of Enum.
It's few hours work with testing and watching new series of GoT in parallel :).

thomasfredericks, 2019-04-18:
Hi, there is already an `#ifdef` in `Bounce.h` at line 184:
```
#if defined(ARDUINO_STM_NUCLEO_F103RB) || defined(ARDUINO_GENERIC_STM32F103C)
        pinMode(pin, (WiringPinMode)mode);
#else
        pinMode(pin, mode);
#endif
```
Could you please add your test case and see if it compiles?

thomasfredericks, 2019-04-18:
Maybe a more generic `#if defined` would be better.