thomasfredericks/Bounce2 #53: Missing public accessor to pin.

matkor, 2019-09-20:
Please add public accessor to this->pin like:

```
    uint8_t get_pin(){
      return this->pin;
    }
```

TIA

thomasfredericks, 2019-09-24:
I will have to think about this. A more abstract class has often been requested allowing debouncing of more than just a single hardware pin. So I will not implement this right now, but later. I need to make a base class that is extended to hardware pins.

matkor, 2019-09-24:
Cool.

For now I can live with:
`class BounceExt: public Bounce
{
  public:
    BounceExt(uint8_t pin, uint16_t interval_millis ) :
      Bounce(pin, interval_millis) {};
    uint8_t get_pin() {
      return pin;
    }
};`

It allows save byte of RAM each time Bounce is member of class, when pin must be later used in class.

septillion-git, 2019-09-29:
Why does that save RAM? Because in most situations you can have a const in your code which you pass to Bounce2 aka it takes no space in RAM.

matkor, 2019-09-30:
> (...) in most situations you can have a const in your code which you pass to Bounce2 aka it takes no space in RAM.

Bounce objects store pin number as their attribute, so most likely byte of RAM is used regardless how init value was passed.

> Why does that save RAM?

I do not have to duplicate pin number data like:

```  c++
class PushButton{
    Bounce debouncer;
    uint8_t pin;   // duplicate value as in debouncer.pin

```

jamesmyatt, 2020-07-21:
@matkor, why do you store the debouncer and pin separately?

matkor, 2020-07-22:
> @matkor, why do you store the debouncer and pin separately?

I do not.
I just need access pin having only Bounce instance.

thomasfredericks, 2020-08-07:
I merged a new version of the library where I split the hardware layer from the debouncing algorithm. It should now be easier to make custom classes.

@matkor We could now add a getPin function to the Bounce class.

thomasfredericks, 2021-06-28:
Was added to v2.58

closed 2021-06-28 by thomasfredericks