thomasfredericks/Bounce2 #54: Feature request: .changed()

septillion-git, 2019-09-29:
No idea why I didn't implement this is the past when I made changes to Bounce2 but why only let .update() return if it changed. There are situations you don't care if it's a .fell() or a .rose() but now you have to change both (which is even extra work because it's made out the current state and the fact if it changed) to simply see if the pin changed.

thomasfredericks, 2019-09-29:
I am sorry, I don't understand.

septillion-git, 2019-09-29:
Sorry, should have explained it a bit more. But in short 👼 :
```C++
bool Bounce::changed()
{
    return getStateFlag(CHANGED_STATE);
}
```

thomasfredericks, 2019-10-05:
Ok, I added a protected `changed()` function in the testing branch: https://github.com/thomasfredericks/Bounce2/tree/Testing

Tell me if it is what you want and if it works.

septillion-git, 2019-10-06:
I don't know why you want it protected? Just have it public 😃

thomasfredericks, 2019-10-06:
Oups

MartinNel, 2019-11-21:
The changed `function() `in the header uses a literal `CHANGED_STATE`, which is not defined in the header file. This gives a compilation error. I suggest moving the literal(s) from the H to the CPP file.

thomasfredericks, 2019-11-22:
Thank you @MartinNel
Fixed

closed 2019-11-22 by thomasfredericks