thomasfredericks/Bounce2 #62: State methods

jamesmyatt, 2020-07-21:
Why are the following methods all private:

```
  private:
    inline void changeState();
    inline void setStateFlag(const uint8_t flag)    {state |= flag;}
    inline void unsetStateFlag(const uint8_t flag)  {state &= ~flag;}
    inline void toggleStateFlag(const uint8_t flag) {state ^= flag;}
    inline bool getStateFlag(const uint8_t flag)    {return((state & flag) != 0);}
```

when the attribute that they concern is protected?

```
 protected:
    uint8_t state;
```

I'm not sure it makes sense to be more restrictive of the methods than the underlying attribute.

thomasfredericks, 2020-07-24:
This is just an oversight. I will look into this when I come back from vacation. I want to split the debouncing code from the hardware layer at the same time.

thomasfredericks, 2020-08-07:
@jamesmyatt Those methods are private because they might change in the future. They are linked to space saving and not to the functioning of the algorithm. Anyways, I merged a lot of changes.

closed 2020-08-07 by thomasfredericks