thomasfredericks/Bounce2 #69: Request:  add Button class state output method isPressed()

EclipseBoom, 2020-08-25:
The Button class doesn't have a logically correct .read() method that takes the PressedState into account.  The pressed() and released() methods are one-time changed state outputs.  Sometimes I want to detect if a Button is being held down.

Something like isPressed() or isReleased() would make logical sense.

Thanks for your work on this library.  So simple yet so useful.

thomasfredericks, 2020-08-25:
Hi, the read returns the electrical state of the pin. This does not change with a pull-up or pull-down button circuit.

Does this merge meet your other needs? https://github.com/thomasfredericks/Bounce2/pull/67

closed 2020-08-25 by thomasfredericks

EclipseBoom, 2020-08-25:
Yes I understand the method from Bounce returns the electrical state of the pin.  However, this could be opposite of the logical state of being pressed if the pressed state as LOW electrically which is the most common situation (and NOT the default BTW).  Read() gives LOW=false when the button is pressed in the majority of cases.  This basically makes the Button class useless because you can't use read() without knowing explicitly the electrical sense of the button.  So if you have to know the physical connection explicitly why bother with the Button class at all?  If you can abstract the electrical state by PressedState() in your definition section all your code will stay correct and abstracted if the hardware changes in the future which is, as I see it, the only real reason for Button in the first place.