thomasfredericks/Bounce2 #70: Hardware abstraction should be by composition rather than inheritance

jamesmyatt, 2020-08-26:
Especially since there is a Button class now, the hardware abstraction should be done by composition rather than inheritance. For example, it's not possible to define a Button instance that uses something other than digitalRead to read the state. What do you think?

thomasfredericks, 2020-08-27:
Hum, interesting. But how would this be implemented?

KenwoodFox, 2021-12-21:
I also want to find a way to do this, id like to be able to define a button instance with an input that is not digitalRead, maybe add in an override?

thomasfredericks, 2021-12-22:
@KenwoodFox this can already be implemented. See the source for `class Bounce` in `Bounce2.h`.

You can set your custom override for digitalRead :
`virtual bool readCurrentState() { return digitalRead(pin); }`

KenwoodFox, 2021-12-24:
> @KenwoodFox this can already be implemented. See the source for `class Bounce` in `Bounce2.h`.
> 
> You can set your custom override for digitalRead : `virtual bool readCurrentState() { return digitalRead(pin); }`

This is pretty awesome! thanks so much.

KenwoodFox, 2021-12-27:
> @KenwoodFox this can already be implemented. See the source for `class Bounce` in `Bounce2.h`.
> 
> You can set your custom override for digitalRead : `virtual bool readCurrentState() { return digitalRead(pin); }`

Sorry to ask, but where would you implement this? Should this be done for every button instance you create or do i need to make my own button class using inheriting from bounce?

Thynix, 2022-02-28:
It isn't a complete replacement because it doesn't have complex debouncing techniques, but the [`Debounce`](https://github.com/apechinsky/arduino-libs/blob/master/Debouncer/examples/basic/sketch.ino) library allows passing in the value to use.