thomasfredericks/Bounce2 #72: Can't compile for Arduino Zero

EliasA97, 2020-12-07:
Hi, your library works very well with arduino due and uno. Unfortunatelly I need to use it on a project involving an arduino zero. It seems that there are some problems and can't compile when a board with SAMD is selected. Is there any chance to fix that?
Thank's!

thomasfredericks, 2020-12-08:
What are your problems?

EliasA97, 2020-12-08:
Hello Thomas, these are the errors the IDE gives me. I don't know if it helps but I don't have another way to show you the error.

`Arduino: 1.8.13 (Windows Store 1.8.42.0) (Windows 10), Πλακέτα:"Arduino Zero (Programming Port)"

sender:29:8: error: 'Button select' redeclared as different kind of symbol

 Button select = Button();

        ^~~~~~

In file included from c:\users\anast\onedrive\�������\arduinodata\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4\arm-none-eabi\include\sys\types.h:68:0,

                 from c:\users\anast\onedrive\�������\arduinodata\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4\arm-none-eabi\include\stdio.h:61,

                 from C:\Users\anast\OneDrive\�������\ArduinoData\packages\arduino\hardware\samd\1.8.9\cores\arduino/Print.h:23,

                 from C:\Users\anast\OneDrive\�������\ArduinoData\packages\arduino\hardware\samd\1.8.9\cores\arduino/Stream.h:26,

                 from C:\Users\anast\OneDrive\�������\ArduinoData\packages\arduino\hardware\samd\1.8.9\cores\arduino/HardwareSerial.h:24,

                 from C:\Users\anast\OneDrive\�������\ArduinoData\packages\arduino\hardware\samd\1.8.9\cores\arduino/Arduino.h:78,

                 from sketch\sender.ino.cpp:1:

c:\users\anast\onedrive\�������\arduinodata\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4\arm-none-eabi\include\sys\select.h:73:5: note: previous declaration 'int select(int, _types_fd_set*, _types_fd_set*, _types_fd_set*, timeval*)'

 int select __P ((int __n, fd_set *__readfds, fd_set *__writefds,

     ^~~~~~

C:\Users\anast\OneDrive\Bell Controller\Bell Controller CPU remote v.3\sender\sender.ino: In function 'void setup()':

sender:92:10: error: request for member 'attach' in 'select', which is of non-class type 'int(int, _types_fd_set*, _types_fd_set*, _types_fd_set*, timeval*)'

   select.attach(SELECT, INPUT_PULLUP);

          ^~~~~~

sender:93:10: error: request for member 'interval' in 'select', which is of non-class type 'int(int, _types_fd_set*, _types_fd_set*, _types_fd_set*, timeval*)'

   select.interval(5);

          ^~~~~~~~

sender:94:10: error: request for member 'setPressedState' in 'select', which is of non-class type 'int(int, _types_fd_set*, _types_fd_set*, _types_fd_set*, timeval*)'

   select.setPressedState(LOW);

          ^~~~~~~~~~~~~~~

C:\Users\anast\OneDrive\Bell Controller\Bell Controller CPU remote v.3\sender\sender.ino: In function 'void loop()':

sender:193:10: error: request for member 'update' in 'select', which is of non-class type 'int(int, _types_fd_set*, _types_fd_set*, _types_fd_set*, timeval*)'

   select.update();

          ^~~~~~

C:\Users\anast\OneDrive\Bell Controller\Bell Controller CPU remote v.3\sender\sender.ino: In function 'void mainMenu()':

sender:299:21: error: request for member 'pressed' in 'select', which is of non-class type 'int(int, _types_fd_set*, _types_fd_set*, _types_fd_set*, timeval*)'

   } else if (select.pressed()) {

                     ^~~~~~~

exit status 1

'Button select' redeclared as different kind of symbol

`

thomasfredericks, 2020-12-08:
This does not seem to be a problem with Bounce2. You seem to be overriding a previous definition of select.

`sys\select.h:73:5: note: previous declaration 'int select(int, _types_fd_set*, _types_fd_set*, _types_fd_set*, timeval*)'`

The simplest is to use another name than select.

EliasA97, 2020-12-08:
That's weird. Then why this problem doesn't occure when choosing another non SAMD arduino board?

EliasA97, 2020-12-08:
It's very weird kind of problem. I just renamed my 'select' button to 'sel' and it now compiles.
Anyway, thank's for answering!

closed 2020-12-08 by EliasA97