thomasfredericks/Bounce2 #75: Library works on arduino Due but not on Zero (same sketch).

EliasA97, 2021-02-05:
Hello, the title says it all. My code works fine when I upload it on the Due, but if I'll try to compile it for the Zero (SAMD21) I get these errors:

`
Remote_Controller_v3:50:17: error: 'Bounce2::Button select' redeclared as different kind of symbol
In file included from c:\users\anast\onedrive\�������\arduinodata\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4\arm-none-eabi\include\sys\types.h:68:0,
                 from c:\users\anast\onedrive\�������\arduinodata\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4\arm-none-eabi\include\stdio.h:61,
                 from C:\Users\anast\OneDrive\�������\ArduinoData\packages\arduino\hardware\samd\1.8.11\cores\arduino/api/Print.h:22,
                 from C:\Users\anast\OneDrive\�������\ArduinoData\packages\arduino\hardware\samd\1.8.11\cores\arduino/api/ArduinoAPI.h:31,
                 from C:\Users\anast\OneDrive\�������\ArduinoData\packages\arduino\hardware\samd\1.8.11\cores\arduino/Arduino.h:23,
                 from sketch\Remote_Controller_v3.ino.cpp:1:
c:\users\anast\onedrive\�������\arduinodata\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4\arm-none-eabi\include\sys\select.h:73:5: note: previous declaration 'int select(int, _types_fd_set*, _types_fd_set*, _types_fd_set*, timeval*)'
 int select __P ((int __n, fd_set *__readfds, fd_set *__writefds,
     ^~~~~~
C:\Users\anast\OneDrive\Υπολογιστής\Remote controller v.3\Remote_Controller_v3\Remote_Controller_v3.ino: In function 'void setup()':
Remote_Controller_v3:171:10: error: request for member 'attach' in 'select', which is of non-class type 'int(int, _types_fd_set*, _types_fd_set*, _types_fd_set*, timeval*)'
Remote_Controller_v3:172:10: error: request for member 'interval' in 'select', which is of non-class type 'int(int, _types_fd_set*, _types_fd_set*, _types_fd_set*, timeval*)'
Remote_Controller_v3:173:10: error: request for member 'setPressedState' in 'select', which is of non-class type 'int(int, _types_fd_set*, _types_fd_set*, _types_fd_set*, timeval*)'
C:\Users\anast\OneDrive\Υπολογιστής\Remote controller v.3\Remote_Controller_v3\Remote_Controller_v3.ino: In function 'void loop()':
Remote_Controller_v3:235:10: error: request for member 'update' in 'select', which is of non-class type 'int(int, _types_fd_set*, _types_fd_set*, _types_fd_set*, timeval*)'
Remote_Controller_v3:270:85: error: request for member 'pressed' in 'select', which is of non-class type 'int(int, _types_fd_set*, _types_fd_set*, _types_fd_set*, timeval*)'
C:\Users\anast\OneDrive\Υπολογιστής\Remote controller v.3\Remote_Controller_v3\Remote_Controller_v3.ino: In function 'void mainMenu()':
Remote_Controller_v3:305:21: error: request for member 'pressed' in 'select', which is of non-class type 'int(int, _types_fd_set*, _types_fd_set*, _types_fd_set*, timeval*)'
exit status 1
'Bounce2::Button select' redeclared as different kind of symbol
`

thomasfredericks, 2021-02-05:
I can't tell for sure as you just dumping the error with no context, but you seem to be using a variable name that is already declared.

closed 2021-02-05 by thomasfredericks

EliasA97, 2021-02-05:
I don't think the the problem is naming. If so, then why it runs fine on the Due while giving problems on the Zero?

thomasfredericks, 2021-02-05:
Again, you did not post your code or context hence one can only guess. But the compiler is complaining that `select` is being redefined. You also ALREADY SUBMITTED AN ISSUE WITH THIS AND I RESOLED IT : https://github.com/thomasfredericks/Bounce2/issues/72