thomasfredericks/Bounce2 #80: Problems with currentDuration ()

fernando-inf, 2021-09-09:
Hi, I have tried using the "`.currentDuration ()`" function from the Bonce2 library, but the best I have achieved is using whit "`.rose`" and "`.isPressed`". However, when you keep the button pressed to go to function 2, function 1 is also activated. In my example, if I want to change only the value of "`toogleState`", I cannot do it without also changing the value of "bank". And as you can see I also had to add a Delay in function 2 and a double condition "`&& pedal_1.currentDuration ()> 500 && pedal_1.currentDuration () <1000`" to avoid problems. 

```
#include <Bounce2.h>

Bounce2::Button pedal_1 = Bounce2::Button();
int bank = 0;
int toogleState = 0;

void setup() {
  Serial.begin(9600);
  pedal_1.attach(2, INPUT_PULLUP);
  pedal_1.interval(25);
  pedal_1.setPressedState(LOW);
}

void loop() {
  pedal_1.update();
  /*FUNCTION 1*/
  if (pedal_1.rose() && pedal_1.currentDuration() < 50) {
    bank++;
    Serial.print("Bank:");
    Serial.println(bank);
  }

  /*FUNCTION 2*/
  else if (pedal_1.isPressed() && pedal_1.currentDuration() > 500 && pedal_1.currentDuration() < 1000) {
    delay(500);
    toogleState++;
    Serial.print("ToogleSate:");
    Serial.println(toogleState);
  }
}
```

fernando-inf, 2021-09-09:
Before I was using "while" which works perfect except for interference noises. When I bring my fingers closer, the push button is activated. 

```

int startTime1 = 0;
int bank = 0;
int toogleState = 0;

void setup() {
  Serial.begin(9600);
  pinMode(2, INPUT_PULLUP);
}
void loop() {
  while (digitalRead(2) == LOW) {
    delay(10);
    startTime1 = startTime1 + 10;
  }

  //FUNCTION 1
  if (startTime1 >= 500) {
    bank++;
    Serial.print("Bank:");
    Serial.println(bank);
  }

  //FUNCTION 2
  else if (startTime1 >= 5) {
    toogleState++;
    Serial.print("ToogleSate:");
    Serial.println(toogleState);
  }
  startTime1 = 0;

}
```

thomasfredericks, 2021-09-10:
When the state changes, the currentDuration is always 0 since the state changed and the currentDuration is 0. You probably want previousDuration.

fernando-inf, 2021-09-10:
Muchas gracias por la pronta respuesta y la sugerencia!. Usted tenía razón, lo solucioné con .previousDuration(). El código quedo así y funciona perfecto:

```
#include <Bounce2.h>
Bounce2::Button pedal_1 = Bounce2::Button();
int bank = 0;
int toogleState = 0;

void setup() {
  Serial.begin(9600);
  pedal_1.attach(2, INPUT_PULLUP);
  pedal_1.interval(25);
  //pedal_1.setPressedState(LOW);
}

void loop() {
  pedal_1.update();
  if ( pedal_1.changed() ) {
    if ( pedal_1.read() == HIGH ) {

     //FUNCTION 2
      if ( pedal_1.previousDuration() > 500 ) {
        toogleState++;
        Serial.print("ToogleSate:");
        Serial.println(toogleState);
      }

     //FUNCTION 1
      else     {
        bank++;
        Serial.print("Bank:");
        Serial.println(bank);
      }
    }
  }
}
```

thomasfredericks, 2021-09-10:
You are welcome.

closed 2021-09-10 by thomasfredericks