thomasfredericks/Bounce2 #82: New release and deprecate duration()

Stutchbury, 2022-01-24:
Hi Thomas - thank you very much for your excellent library.

Could I request a new release and pull to the Arduino Library repository and if possible, to mark ```duration()``` as deprecated? 

Many thanks

thomasfredericks, 2022-01-24:
I thought I might have acted quickly enough as  `duration` was a new fixture. Is `duration` still documented? I have yet to find the right way to mark some functions as deprecated only when used...

Stutchbury, 2022-01-24:
I'm not sure if it was documented or not but it is listed as 'public:' in the v2.60 release that is available via the Arduino Library Manager. 

WRT deprecation markup, I'm not sure either - C++ is not my first language...

Happy to align with your releases - I just got an issue raised on my EncoderButton library, so thought I'd pass it on ;-)

thomasfredericks, 2022-02-04:
Do you need me to readd duration() and mark it as deprecated (as I removed it)? Or can we consider the issue resolved?

Stutchbury, 2022-02-04:
If you can work out how to mark it, that would be good for anyone who uses your latest git with my library release (I suspect a very small pool) but if/when you create a new release for the Arduino Library Manager, I'll update my code & create a new release.

Thank you.

ckxng, 2022-03-19:
Ideally, with [semantic versioning](https://semver.org/), this would be a breaking change and should result in an increase in major version.  Library managers that depend on semantic versioning will "do the right thing" in that case.

For example, some packages are consuming Bounce2 via. platformio using a version string such as `thomasfredericks/Bounce2@^2.60`.  This indicates that the package wants to consume the latest non-breaking change starting with 2.60.  (Where the version is x.y.z, x is fixed while y and z can update.)  Packages who do this (which is the default behavior) and also use the duration function (such as lendres/ButtonSuite and Stutchbury/EventButton) are now more difficult to use.

The workaround is to use something like `thomasfredericks/Bounce2@~2.60`, which would say that the package wishes to receive patches only, no breaking changes _or_ feature improvements.  (Where the version is x.y.z, x and y are fixed while z can update.)  

In my case, I've added `thomasfredericks/Bounce2@~2.60` to my platformio.ini lib_deps directly, which solved the problem for me in the interim.

The cleanest solution, which would align with semver (and therefore cause correct behavior in library managers) would be to wrap Debouncer::currentDuration with Debouncer::duration, in the v2 branch, and if you would like to remove duration entirely - to release that as 3.0.0.

thomasfredericks, 2022-03-21:
Done

Stutchbury, 2022-03-21:
Thanks Thomas