thomasfredericks/Bounce2 #83: Functions do work with the array type of setup

SouthernAtHeart, 2022-02-16:
The example under more called bounceMore looks great as I have lots of buttons.

So I have this:
` //Setup buttons
  for (int i = 0; i < NUM_BUTTONS; i++) {
    buttons[i].attach( BUTTON_PINS[i] , INPUT_PULLUP  );       //setup the bounce instance for the current button
    buttons[i].interval(25);              // interval in ms

  }`

But, I find that I can't use the function .isPressed with it.  Like this: 
buttons[I].isPressed 
so I have to use the individual method for assigning each of my 13 buttons.

SouthernAtHeart, 2022-02-16:
To simplify what I'm saying: this is the exact code from the example.  And the line that says, 
    if ( buttons[i].fell() ) {
you can't use buttons[I].isPressed.  

`// Detect the falling edge of multiple buttons.
// Eight buttons with internal pullups.
// Toggles a LED when any button is pressed.
// Buttons on pins 2,3,4,5,6,7,8,9

// Include the Bounce2 library found here :
// https://github.com/thomasfredericks/Bounce2
#include <Bounce2.h>

#define LED_PIN 13

#define NUM_BUTTONS 8
const uint8_t BUTTON_PINS[NUM_BUTTONS] = {2, 3, 4, 5, 6, 7, 8, 9};

int ledState = LOW;

Bounce * buttons = new Bounce[NUM_BUTTONS];

void setup() {

  for (int i = 0; i < NUM_BUTTONS; i++) {
    buttons[i].attach( BUTTON_PINS[i] , INPUT_PULLUP  );       //setup the bounce instance for the current button
    buttons[i].interval(25);              // interval in ms
  }

  // Setup the LED :
  pinMode(LED_PIN, OUTPUT);
  digitalWrite(LED_PIN, ledState);

}

void loop() {

  bool needToToggleLed = false;

  for (int i = 0; i < NUM_BUTTONS; i++)  {
    // Update the Bounce instance :
    buttons[i].update();
    // If it fell, flag the need to toggle the LED
    if ( buttons[i].fell() ) {
      needToToggleLed = true;
    }
  }

  // if a LED toggle has been flagged :
  if ( needToToggleLed ) {
    // Toggle LED state :
    ledState = !ledState;
    digitalWrite(LED_PIN, ledState);
  }

}
`

thomasfredericks, 2022-02-17:
isPressed is part of the Button class : http://thomasfredericks.github.io/Bounce2/files/class_bounce2_1_1_button.html

Just change `Bounce * buttons = new Bounce[NUM_BUTTONS];` to `Bounce2::Button* buttons = new Bounce[NUM_BUTTONS];`

closed 2022-02-17 by thomasfredericks

SouthernAtHeart, 2022-02-17:
Thanks so much! That makes code smaller