thomasfredericks/Bounce2 #85: Does #define BOUNCE_LOCK_OUT work?

drf5n, 2022-03-26:
I was trying it in Wokwi as 

https://wokwi.com/projects/324855073217708626

and didn't see that it had an effect.

The documentation at https://github.com/thomasfredericks/Bounce2#lock-out-interval says 

> By defining "#define BOUNCE_LOCK_OUT" in "Bounce.h" (or in your code before including "Bounce.h") you can activate an alternative debouncing method. This method is a lot more responsive, but does not cancel noise.

but it does not seem to have an effect.

Output from that script's simulation: 
```
2297ms leading Edge Debounce: 1
2297ms Wasser Debounce: 1
2349ms Bounce2 Debounce: 1
2350ms Arduino Debounce.ino: 1
```

Some discussion at https://forum.arduino.cc/t/leading-edge-debouncing/964407/15

drf5n, 2022-03-27:
I updated the Wokwi Simulation at https://wokwi.com/projects/324855073217708626 to produce a simulated noisy square wave and saw no difference between the #define BOUNCE_LOCK_OUT and not.

With a commented-out `//#define BOUNCE_LOCK_OUT`:

```
500ms leading Edge Debounce: 1
500ms Wasser Debounce: 1
550ms Bounce2 Debounce: 1
551ms Arduino Debounce.ino: 1
1052ms Wasser Debounce: 2
2001ms leading Edge Debounce: 2
2001ms Wasser Debounce: 3
2091ms Bounce2 Debounce: 2
2093ms Arduino Debounce.ino: 2
3001ms leading Edge Debounce: 3
3001ms Wasser Debounce: 4
3093ms Bounce2 Debounce: 3
3094ms Arduino Debounce.ino: 3
4001ms leading Edge Debounce: 4
4001ms Wasser Debounce: 5
4092ms Bounce2 Debounce: 4
4093ms Arduino Debounce.ino: 4
```

With ` #define BOUNCE_LOCK_OUT` uncommented:
```

500ms leading Edge Debounce: 1
500ms Wasser Debounce: 1
550ms Bounce2 Debounce: 1
551ms Arduino Debounce.ino: 1
1052ms Wasser Debounce: 2
2001ms leading Edge Debounce: 2
2001ms Wasser Debounce: 3
2091ms Bounce2 Debounce: 2
2093ms Arduino Debounce.ino: 2
3001ms leading Edge Debounce: 3
3001ms Wasser Debounce: 4
3093ms Bounce2 Debounce: 3
3094ms Arduino Debounce.ino: 3
4001ms leading Edge Debounce: 4
4001ms Wasser Debounce: 5
4092ms Bounce2 Debounce: 4
4093ms Arduino Debounce.ino: 4
```

thomasfredericks, 2022-03-28:
Hello @drf5n 

I tested with the following simpler example : https://wokwi.com/projects/327380656345580114

I tested with the simulator and in reality. With a real circuit everything works. With the simulator, there seems to be a bug with Wokwi where the `#define` is not taken into account. 

To answer you initial question : Yes, #define BOUNCE_LOCK_OUT works.

closed 2022-03-28 by thomasfredericks

drf5n, 2022-03-28:
Thank you.  I raised an issueover on the simulator: https://github.com/wokwi/avr8js/issues/121

drf5n, 2022-03-28:
It isn't just the simulator -- it doesn't do it my Arduino 1.8.19

I tried this copy of bounce_basic.ino on a mega with a 5000ms interval and `#define BOUNCE_LOCK_OUT`

```
/*
  DESCRIPTION
  ====================
  Simple example of the Bounce library that switches a LED when
  a state change (from HIGH to LOW) is triggered (for example when a button is pressed).
  Set BOUNCE_PIN to the pin attached to the input (a button for example).
  Set LED_PIN to the pin attached to a LED.
*/

// WE WILL attach() THE Bounce INSTANCE TO THE FOLLOWING PIN IN setup()
#define BOUNCE_PIN 2

// DEFINE THE PIN FOR THE LED :
// 1) SOME BOARDS HAVE A DEFAULT LED (LED_BUILTIN)
#define LED_PIN LED_BUILTIN
// 2) OTHERWISE SET YOUR OWN PIN
// #define LED_PIN 13

// Include the Bounce2 library found here :
// https://github.com/thomasfredericks/Bounce2
#define BOUNCE_LOCK_OUT
#include "Bounce2.h"

// INSTANTIATE A Bounce OBJECT
Bounce bounce = Bounce();

// SET A VARIABLE TO STORE THE LED STATE
int ledState = LOW;

void setup() {

  // BOUNCE SETUP

  // SELECT ONE OF THE FOLLOWING :
  // 1) IF YOUR INPUT HAS AN INTERNAL PULL-UP
  bounce.attach( BOUNCE_PIN ,  INPUT_PULLUP ); // USE INTERNAL PULL-UP
  // 2) IF YOUR INPUT USES AN EXTERNAL PULL-UP
  //bounce.attach( BOUNCE_PIN, INPUT ); // USE EXTERNAL PULL-UP

  // DEBOUNCE INTERVAL IN MILLISECONDS
  bounce.interval(5000); // interval in ms

  // LED SETUP
  pinMode(LED_PIN, OUTPUT);
  digitalWrite(LED_PIN, ledState);

}

void loop() {
  // Update the Bounce instance (YOU MUST DO THIS EVERY LOOP)
  bounce.update();

  // <Bounce>.changed() RETURNS true IF THE STATE CHANGED (FROM HIGH TO LOW OR LOW TO HIGH)
  if ( bounce.changed() ) {
    // THE STATE OF THE INPUT CHANGED
    // GET THE STATE
    int deboucedInput = bounce.read();
    // IF THE CHANGED VALUE IS LOW
    if ( deboucedInput == LOW ) {
      ledState = !ledState; // SET ledState TO THE OPPOSITE OF ledState
      digitalWrite(LED_PIN,ledState); // WRITE THE NEW ledState
    }
  }

}
```

It works if I edit .../libraries/Bounce2.h and uncomment the #define BOUNCE_LOCK_OUT line, but if I put a #define BOUNCE_LOCK_OUT into my sketch it does not.

If I put an `#error drf5n` into the libraries/Bounce2/src/Bounce2.cpp like:

```
bool Debouncer::update()
{

    unsetStateFlag(CHANGED_STATE);
#ifdef BOUNCE_LOCK_OUT
#error drf5n
    // Ignore everything if we are locked out
    if (millis() - previous_millis >= interval_millis) {
        bool currentState = readCurrentState();
        if ( currentState != getStateFlag(DEBOUNCED_STATE) ) {
            previous_millis = millis();
            changeState();
        }
    }
...
```

defining the flag in the sketch https://github.com/thomasfredericks/Bounce2/blob/master/examples/bounce_basic/bounce_basic.ino doesn't trigger the error

but defining the flag in Bounce2.h does trigger it.

I think the method works, but is not enabled by a #define in the sketch as described in [the docs](https://github.com/thomasfredericks/Bounce2#lock-out-interval) with  "(or in your code before including "Bounce.h") "