thomasfredericks/Bounce2 #92: define bounce in two ways?

billbill100, 2023-05-11:
I'm using an ESP32, which may be causing the issue,

Following the examples, I defined my bounce buttons as:-
' Bounce SampleButton = Bounce(); '
and
' while (SampleButton.fell() == false) '

which worked without issue, but, when changing my code to :-
' if (SampleButton.read() == LOW) '

it errors with 
'Compilation error: 'SampleButton' does not name a type '
Similar issue if I use other keywords, like ' pressed ' 

After hunting all the example, I found a different way to define the bounce buttons in the re-trigger example.
' Bounce2::Button SampleButton = Bounce2::Button(); '
'  SampleButton.setPressedState(LOW); '

I'm very confused.

thomasfredericks, 2023-05-15:
In your first code sample, you are instancing the `Bounce` class.

In your second sample, your are instancing the `Button` class.

They are not the same. The `Button` class is a subclass of the `Bounce` class.

billbill100, 2023-05-15:
Hi,
Thanks for the reply, but that is even more confusing.

I just copied the bounce basic example code to use in my program, but some of the keywords in 'keywords.txt' would not work, so found the example bounce_button.ino, which does allow 'LOW' keyword to be used.

I have no idea what the difference between button and bounce class is? 

I changed my code slightly, as below, to allow SampleButton to use ' SampleButton.setPressedState(LOW); '
However, I have no clue why or if I am doing it correctly.
Thanks.

// name buttons in Bounce2 libary
Bounce2::Button SampleButton = Bounce2::Button();
Bounce isoButton = Bounce();
//Bounce SampleButton = Bounce(); // removed this line and added top line.
Bounce UpButton = Bounce();
Bounce DownButton = Bounce();
Bounce encoderButton = Bounce();

thomasfredericks, 2023-05-15:
The **Issues** section of this GitHub are for problems related to the library and with problems found with the unmodified examples.
We cannot provide help for general coding.
Here is the description of the classes taken from the first page of the documentation:
```
The library is composed of three classes:

    Debouncer : The code that does the actual debouncing. Only advanced users should play with this class.
    Bounce : This is the general use library. It links the Debouncer to a hardware pin on your board.
    Button : A special version of Bounce for buttons that are pressed.
```
Please consult the documentation and examples to learn how to use each class. For example, [here is the documentation ](http://thomasfredericks.github.io/Bounce2/files/class_bounce2_1_1_button.html#a3c88b7938b26bca9dc2c7e72aedc442e)for  `setPressedState()`.

thomasfredericks, 2023-05-15:
**Keywords do not function as documentation.** They do not provide scope and are not intended to be read by humans. Not all classes support all keywords. Do not use them as a guide. 

Use the examples and documentation: 
http://thomasfredericks.github.io/Bounce2/files/index.html

As stated on the main documentation page, please post your questions here:
[image]
Here is the link : http://forum.arduino.cc/index.php?topic=266132.0

closed 2023-05-15 by thomasfredericks