thomasfredericks/Bounce2 #95: Question regarding multiple button instances in a switch statement.

Fusionlight13, 2024-05-27:
I made a function with a switch statement to call each button by value to update and return the .pressed method. Multiple presses are returning true which is really confusing me. I use the INPUT_PULLUP mode and attach the wires to ground and the digital pins. I can't figure out if it's an issue with my wiring or the library.
 ```python
bool buttonClicked(int id)
{
  bool buttonBool;
  switch(id)
  {
    case 1:
      button1.update();
      buttonBool = button1.pressed();
      break;
    case 2:
      button2.update();
      buttonBool = button2.pressed();
      break;
    case 3:
      button3.update();
      buttonBool = button3.pressed();
      break;
  }
  return buttonBool;
}
```
```python
void loop() {
  // put your main code here, to run repeatedly:
  if(buttonClicked(1))
  {
    changeLed();
  }
    else if(buttonClicked(2))
  {
    
    Serial.println("Button 2 has been pressed!");
  }
  else if(buttonClicked(3))
  {
    Serial.println("Button 3 clicked!");
  }
}
```

thomasfredericks, 2024-05-28:
It is probably your code. The `else if `statements in` loop() `do not make sense in this context and should be replace with separate `if` statements. Otherwise the button updates are not consistently executed every loop.

If you want an `else if structure`, run all updates first in a separate block and then check if they were pressed in a separate `else if `block.

thomasfredericks, 2024-05-28:
For further issues on general coding please use https://forum.arduino.cc/latest

closed 2024-05-28 by thomasfredericks

Fusionlight13, 2024-05-28:
Thanks for the help!